mciantyre/teensy4-rs #4: teensy4-rt is unusable without PAC

mciantyre, 2019-11-17:
Unlike the `cortex-m-rt` runtime crate, which is usable without any other peripheral access crate (PAC), the `teensy4-rt` implicitly depends on the `imxrt1060-pac`. Specifically, it depends on the `imxrt1060-core` crate, which describes the interrupt table and available interrupt enums. For those familiar with `cortext-m-rt`, it's as if we're always using the `"cortext-m-rt/device"` feature.

Refactor the `teensy4-rt` crate to be usable without the `imxrt1060` crates. It will serve value for developers who need a minimal runtime for the Teensy 4, and who are not interested in learning / using the peripheral APIs exported by `svd2rust`.

A solution should mimic the behavior of the `cortex-m-rt` crate, which will provide a vector table with all interrupts registered to the default handler. This falls under the umbrella of maintaining compatibility with the `cortex-m-rt` crate.

Shock-1, 2020-01-05:
where's the teensy4-rt crate?

mciantyre, 2020-01-05:
It was renamed to `imxrt1062-rt` as part of #17. I updated the issue description and title. Thanks for bringing this up!

closed 2020-10-16 by bors[bot]