mciantyre/teensy4-rs #37: undefined symbol __svectors

drewhepp, 2020-01-27:
Hello,

I'm attempting to compile a simple usb logger as a test, but when I target thumbv7em-none-eabihf, I get a linker error "undefined symbol __svectors".

I assume this has something to do with me messing up the toolchain install? Any insight as to where I've gone wrong would be appreciated. I have the gcc-arm-none-eabi toolchain on my path.

Interestingly, running "./hardware-test.sh usb" works great.

I can provide more info if necessary. Thanks for all the work on this project, super excited to use it!

drewhepp, 2020-01-27:
After digging around a bit more, I added a line from the .cargo/config for the examples:

[target.thumbv7em-none-eabihf]
rustflags = [
    "-C", "link-arg=-Tlink.x",
]

Which is keeping the linker happy

closed 2020-01-27 by drewhepp

mciantyre, 2020-01-28:
Good catch! Adding the linker script would do it.

I've put together a `cargo generate` template for teensy4-rs projects. The template automatically defines the `.cargo/config` file. Check it out here: https://github.com/mciantyre/teensy4-rs-template