mciantyre/teensy4-rs #49: Collaboration around i.mx rt

bfrog, 2020-02-12:
Hi, this is a fantastic set of work. I'm looking to build software in rust around the nxp evk boards, specifically the 1062 and 1011 evk boards. I'm wondering if a collaborate effort can be done on some common things.

I tried creating a mega pac using svd2rust and found the compile super slow and found this crate noting you had the same issue. I think an alternative to svd2rust using stm32ral's py script and macros might be significantly saner to work with, and started building things around that idea at https://github.com/imxrt-rs/imxrt-rs. The equivalent of the PAC crate, for almost all of the nxp imxrt series chips, builds in *3 seconds*.

It does use macros rather than the type state functions svd2rust generates, but otherwise supports all of the same high level features and rust embedded ecosystem.

I'd really like to try porting your hal to this, and expanding it to support other imxrt chips. I would love to get your approval of that, and perhaps if it pans out, we can work together further on it together.

Thanks!

mciantyre, 2020-02-14:
I'm happy to support a port of the HAL to a RAL-based implementation. I've been struggling with the  PAC's long compile times (see #48). Deep down, I've wanted an alternative to what svd2rust produced, but hadn't looked for alternatives. Thanks for introducing me to the RAL idea.

Let me know if I can help with the HAL port. I might take the opportunity to clean up some HAL internals, and make the peripheral APIs more consistent.

Any thoughts on adopting the FCB generation and runtime crates? They both assume an iMXRT1062, but with some work they should be usable across iMXRT chip variants. I can contribute them to the imxrt-rs org if you find them useful. The runtime crate might be necessary until we have something like rust-embedded/cortex-m-rt#164 available.

I also have some SVD modifications I might be able to formalize in svdtool patches. Thanks for introducing me to that tool, too.

bfrog, 2020-02-14:
Yes, the compile times are painful for pacs, this series just happens to be massive and I think pushes the svd2rust method really outside the realm of usable unfortunately. At least without major improvements to the compiler.

First of all I'm glad to hear your happy to support this.

I'd love help with the hal port, I've started by porting the ccm peripheral, the next up I think would be gpio, so I can get blinky leds everywhere. Past that, its really supporting the rest of the peripherals. This chip has a lot, and the work on USB in rust is especially exciting! 

Any modifications you've done with the SVDs would almost certainly be welcome as svdtool type patches. There's certainly some quirks in there to sort out.

The way the imxrt-ral crate works for each chip is with a feature flag and then creating aliases at the root module. So with a feature flag imxrt_ral::ccm might be for the 1011 or the 1062. This is how I'm planning on writing a hal, this should let me reuse large swaths of code very quickly.

Maybe the same method can be used for your fcb and rt crates in a way that they can support other chips with feature flags taking advantage of any common functionality which there should be quite a bit. 

I'm almost always on in #rust-embedded:matrix.org where many of the other rust embedded authors also chat if you want to discuss anything more readily.

closed 2020-02-18 by bfrog