mciantyre/teensy4-rs #51: Support SPI DMA transfers

dflemstr, 2020-03-02:
I love the recent development and buzz around SPI support. In my application, I want to transfer a lot of data, and it would hence be useful to support DMA transfers for SPI!

mciantyre, 2020-03-11:
Happy to help out. Just want to note that we're re-implementing this repository's HAL on a different foundation. Check out the work in the [imxrt-rs](https://github.com/imxrt-rs/imxrt-rs) repository. #49 talks about the compilation speed improvements gained from using a different register access layer. If we want to implement DMA channels and abstractions, we'll want to add them over in that repo.

Any thoughts about what a DMA API might look like for your use-case? Haven't seen many DMA-based Rust drivers, though I haven't looked too hard.

dflemstr, 2020-03-14:
I have another side project where I've been trying to eek out the maximum amount of performance from SPI in a device independent manner.  I defined the API to look like this: https://github.com/dflemstr/embedded-platform/blob/master/src/spi.rs

Of course, that introduces async as well, while this crate/embedded_hal is very focused on blocking APIs. However the main take-away from that API I think is:

  1. You initiate a transfer which takes ownership of the slice of memory you want to send
  2. You can poll that transfer to completion (and do that in a tight loop if you want a "blocking" API)

mciantyre, 2020-04-07:
I moved this to imxrt-rs/imxrt-rs#43. Let's have follow-up discussions there.

closed 2020-04-07 by mciantyre