mciantyre/teensy4-rs #59: Digital InputPin: configuring pullup/pulldown

Walther, 2020-04-30:
_I'm not 100% sure if this issue should be here or in the `imxrt-rs` crate - feel free to tell me to move this issue there, if it's appropriate!_

Adapted from the `led.rs` example:

```rust
extern crate panic_halt;

use bsp::rt::entry;
use teensy4_bsp as bsp;

use bsp::hal::gpio::IntoGpio;
use embedded_hal::digital::v2::InputPin;
use embedded_hal::digital::v2::OutputPin;

#[entry]
fn main() -> ! {
    let mut peripherals = bsp::Peripherals::take().unwrap();
    let switch = peripherals.pins.p0.into_gpio();
    let mut led = bsp::configure_led(&mut peripherals.gpr, peripherals.pins.p13);

    loop {
        match switch.is_high() {
            Ok(true) => {
                led.set_high().unwrap();
            }
            Ok(false) => {
                led.set_low().unwrap();
            }
            Err(_e) => (),
        }
    }
}
```

This kinda works! I can start the Teensy with the led in off state, and using the physical switch I have (and a resistor) I can switch the led on. Yay! However, I'm using a spst switch, and the pin doesn't reset when I turn the switch back off, because by default there's no pullup/pulldown configured for the input pin. I can manually pull it by jumping it to a GND.

Is configuring a pullup/pulldown supported?

```rust
use bsp::hal::iomuxc::pin_config::pull_up;
peripherals.pins.p0.set_pull_up(pull_up::Values::PullDown_100KOhm).into_gpio();
// no method named `set_pull_up` found for struct `imxrt_hal::iomuxc::gpio::ad_b0::GPIO_AD_B0_03<imxrt_hal::iomuxc::Alt5>` in the current scope

peripherals.pins.p0.into_gpio().set_pull_up(pull_up::Values::PullDown_100KOhm);
// no method named `set_pull_up` found for struct `imxrt_hal::gpio::GPIO1IO03<imxrt_hal::gpio::GPIO1, imxrt_hal::gpio::Input>` in the current scope
```

---

This might sound like a specific little case, and it is highly likely I've just missed something, being relatively new to embedded Rust. But maybe at least something could be improved about pin usage documentation and/or ergonomics? I think after blinking an onboard led, the next step for a lot of beginners is trying to use pins for interfacing with switches, buttons, more leds, etc.

mciantyre, 2020-04-30:
It should be supported! Check out [the `pin_config` module](https://github.com/imxrt-rs/imxrt-rs/blob/1b9e731f2664877965c04365fc674e99600734f7/imxrt-hal/src/iomuxc/pin_config.rs) in the `imxrt-hal` crate for the full capabilities.

All pins support [a `configure()` method](https://github.com/imxrt-rs/imxrt-rs/blob/1b9e731f2664877965c04365fc674e99600734f7/imxrt-hal/src/iomuxc/macros.rs#L189). Supply a `&PinConfig` that describes your settings. Note that you have to call this on the pin object, before calling `into_gpio()`.

Here's your example, still compiling, but setting a 100KOhm pull-up on the switch. Note the `const SWITCH_PIN_CONFIG` and the call to `configure()`.

```rust
#![no_std]
#![no_main]

extern crate panic_halt;

use bsp::rt::entry;
use teensy4_bsp as bsp;

use bsp::hal::gpio::IntoGpio;
use embedded_hal::digital::v2::InputPin;
use embedded_hal::digital::v2::OutputPin;

use bsp::hal::iomuxc::pin_config::{
    PinConfig,
    PullUp,
};

/// NEW!
///
/// Also, this doesn't have to be `const`; it could be a `let` binding.
const SWITCH_PIN_CONIFG: PinConfig = PinConfig::with_none().set_pull_up(PullUp::PullUp_100KOhm);

#[entry]
fn main() -> ! {
    let mut peripherals = bsp::Peripherals::take().unwrap();
    // Set my pin configuration
    peripherals.pins.p0.configure(&SWITCH_PIN_CONIFG);
    let switch = peripherals.pins.p0.into_gpio();
    let mut led = bsp::configure_led(&mut peripherals.gpr, peripherals.pins.p13);

    loop {
        match switch.is_high() {
            Ok(true) => {
                led.set_high().unwrap();
            }
            Ok(false) => {
                led.set_low().unwrap();
            }
            Err(_e) => (),
        }
    }
}
```

mciantyre, 2020-05-03:
Closing; feel free to re-open if you have any follow-ups.

closed 2020-05-03 by mciantyre

canselcik, 2024-05-27:
Is this still supposed to work the same way?

mciantyre, 2024-05-27:
We've evolved these packages, so the examples above won't work precisely the same way. But we haven't evolved too much. I pushed some commits ([hardware example](https://github.com/mciantyre/teensy4-rs/commit/52e9a7a4606e002ae1140c04e93ef7e89a6577be), [docs](https://github.com/mciantyre/teensy4-rs/commit/e07527192e8475e0552926c2e41b52ffb2ca51b4)) that show how to configure pull ups / downs with today's packages.