mciantyre/teensy4-rs #76: Reprogramming, then quickly WFI-ing, resets Teensy

mciantyre, 2020-08-29:
**Describe the bug**
When you reprogram the Teensy, there's a chance that the chip will reset again when you hit `WFI`. You'll notice that the Teensy's red, programming LED is on, and nothing happens.

**To Reproduce**
In the `pit.rs` example, remove the spin-loop:

https://github.com/mciantyre/teensy4-rs/blob/b9f297e350d2412fdc1b70f5d9f23ee2bcad211a/examples/pit.rs#L66-L69

Then, build and load the example:

```bash
cargo objcopy --example pit --release --no-default-features -- -O ihex target/thumbv7em-none-eabihf/release/examples/pit.hex
teensy_loader_cli -w -v --mcu=TEENSY40 target/thumbv7em-none-eabihf/release/examples/pit.hex
```

Or, use this minimal example:

```rust
#![no_std]
#![no_main]

extern crate panic_halt;

use cortex_m::asm::wfi;
use teensy4_bsp::rt::entry;

#[entry]
fn main() -> ! {
    loop {
        wfi();
    }
}
```

**Expected behavior**
When `main()` is called, I can immediately `WFI` without issues. I can re-program the Teensy with the same example, and it works as expected.

**Additional context**
Work arounds:

- Add a wait, as shown in the `pit.rs` example. A few hundreds of milliseconds seems to be OK. Or, add a busy-loop, as shown in the RTIC examples:

    https://github.com/mciantyre/teensy4-rs/blob/b9f297e350d2412fdc1b70f5d9f23ee2bcad211a/examples/rtic_uart_log.rs#L143-L152

- Power-cycle the Teensy after reprogramming it. There's no issue when booting from a cold start.

You might not notice this when using the Teensy core libraries, since there's [a 300ms delay before calling `main()`](https://github.com/PaulStoffregen/cores/blob/b1ba3875803219a13df7ebfd641b3ddfd9689240/teensy4/startup.c#L138). We could introduce a similar workaround in the runtime crate.

mciantyre, 2020-08-29:
Previous discussions in #64.

ergpopler, 2020-09-24:
I am having this issue when trying to do the instructions on the readme, i cannot get it to work at all!

mciantyre, 2020-09-24:
Thanks for the feedback. I updated the example in the template project so it won't be the first thing new users hit.

Did any of the workarounds described in the issue help?

ergpopler, 2020-09-25:
Yes, it was my bad, when I put in the workaround and compiled it didn't update in my teensy loader, tht was my bad.

mciantyre, 2020-09-26:
No problem! Good to know a manual workaround helped.

I'm exploring a software workaround in #79.

mciantyre, 2020-10-03:
There's a workaround in #79. Leaving this open as my reminder to document it in the runtime crate.

commit ce71c2f2 (2020-10-15): Update documentation

This documents the runtime changes required to address #76.
Closes #76.
closed 2020-10-16 by bors[bot]