mciantyre/teensy4-rs #85: Unable to reproduce the usb example in independant crate

Artefaritaj, 2020-11-30:
### Quick description of the problem

I am trying to test the usb examples on my Teensy 4.1. They are working perfectly fine when compiled from this repo as an example (led is blinking, messages are displayed in the terminal).
But if I want to do the same in a new crate, messages are not displayed (but the LED is blinking).

### How to reproduce

Generate a new project:
```
cargo generate --git https://github.com/mciantyre/teensy4-rs-template --name usbtest
```
Copy the usb.rs from the examples
```
cp teensy4-rs/examples/bsp/src/bin/usb.rs usbtest/src/main.rs
```

Update the Cargo.toml to match the example's one
```
[package]
name = "bsp"
version = "0.1.0"
publish = false
authors = ["Ian McIntyre <ianpmcintyre@gmail.com>"]
edition = "2018"

[dependencies.teensy4-bsp]
version = "0.1"
features = ["rt", "usb-logging"]
path = "../teensy4-rs" #may be removed to use the crates.io version instead

[dependencies]
cortex-m = "0.6.2"
cortex-m-rt = "0.6.13"
embedded-hal = "0.2.4"
log = "0.4.8"
nb = "0.1.2"
panic-halt = "0.2.0"
```

Build
```
cargo objcopy --release --bin usb -- -O ihex usb.hex
```
and program.

### How the bug can be observed

The LED is blinking every 5s as it should, but no messages are displayed on the serial over usb console. Messages can be observed when compiling directly from the examples/bsp directory.

### What I tried

Since the code is identical, the problem is probably in the Cargo.toml. I tried to modify the features for the teensy4-bsp crate (adding and removing "usb-logging") but nothing changes. Looking in the source, "usb-logging" is enabled by default.

I feel I am missing something obvious, but what ?

mciantyre, 2020-11-30:
Reproduced the issue following your steps. Thanks for the great report!

Could you remove this line from the `LoggingConfig` struct, and see if that helps?

https://github.com/mciantyre/teensy4-rs/blob/29e03e935b12ffe4c3798b7bd405c094e58897a2/examples/bsp/src/bin/usb.rs#L25

That `usb` example will _only_ show log messages that are emitted from the `"usb"` module. Since we've changed the name of our program from "usb" to "usbtest," our module name doesn't match the logging filter. That's why we don't see any output.

After removing that line, you should see the log messages from the program. Before the main loop, you may also see debug-level log messages that have to do with clock configuration in the HAL. To filter those out, add a filter with the name `"usbtest"`.

Artefaritaj, 2020-12-01:
Thank you ! It works perfectly as described. May I suggest to add a comment in the usb example :
```
filters: &[("usb", None)], // "usb" to denote that we are keeping only messages originating from the "usb" crate/binary.
```

closed 2020-12-01 by Artefaritaj

arrisde, 2021-03-21:
On a related note, I had the same issue with the "Getting Started" example on the project starting page. Turns out for some reason my module identifies as `hello_world` instead of `hello-world`, so I had to change to
```Rust
const FILTERS: &'static [bsp::usb::Filter] = &[
    ("hello_world", None),
];
```
in `logging.rs` accordingly to see the log output.

mciantyre, 2021-03-21:
Sorry for the trouble. I removed the default filter from the template project to avoid confusion.

The issue was a `{{package-name}}` [cargo generate template](https://github.com/cargo-generate/cargo-generate#templates) that should have been `{{crate_name}}`.