mciantyre/teensy4-rs #95: Remove USB driver

mciantyre, 2021-03-20:
The [`imxrt-usbd` package](https://github.com/imxrt-rs/imxrt-usbd) provides a USB driver that's compatible with the `usb-device` Rust ecosystem. Eventually, it should be sufficient to replace the BSP's custom USB stack. This task tracks the removal of the BSP's USB driver.

mciantyre, 2022-01-02:
Another way to frame this effort: a [`log` implementation](https://crates.io/crates/log) (or [`defmt` implementation](https://defmt.ferrous-systems.com)) that works with [`usbd-serial`](https://crates.io/crates/usbd-serial). This general package could be a win for other embedded Rust users. I've seen hints of people working on this kind of package in chats and issues, but I haven't found working code. Let me know if you know about something like this.

I'd happily contribute, and integrate it in the BSP. Since it would work beyond this BSP, I would recommend maintaining it elsewhere.

brandonros, 2022-05-05:
Any docs on how to attach to device and see the `bsp::usb` logs included with template?

brandonros, 2022-05-05:
```
//! Success criteria: you see log messages when connecting
//! to the Teensy 4 using PuTTY of another serial console.
//! Baud rate is 9600.
```

jspngh, 2022-10-31:
Hi @mciantyre, this seemed like a nice thing to have so I gave it a shot.
I've mainly borrowed from `imxrt-usbd` + `imxrt-uart-log` and haven't found a way to make it HAL-agnostic yet.
Some feedback and ideas for improvements would be appreciated.
You can find the repo here: https://gitlab.com/jspngh/imxrt-usb-log

mciantyre, 2022-10-31:
Thanks for giving this a go. It's great to see this support for today's BSP and USB driver. I'll take a look through to see what kind of uses you're considering.

I'm sorry, I've haven't kept this issue up to date. I've been working on [another logging solution](https://github.com/mciantyre/imxrt-hal/tree/master/logging) in [my imxrt-hal fork](https://github.com/mciantyre/imxrt-hal). This other approach requires a whole change in the imxrt-rs ecosystem, so it'll take longer to realize and would only be usable in a future Teensy 4 BSP.

jspngh, 2022-10-31:
No problem, it looks like a more powerful and flexible solution, I'll keep an eye on your work!
If anything, my attempt served as a learning process.
But maybe someone else finds it useful in the short term, so I wanted to share it.

PR #125 (2023-01-06): Update teensy4-bsp, teensy4-pins to latest imxrt-rs dependencies
This PR updates

- teensy4-bsp to use imxrt-hal 0.5
- teensy4-pins to use imxrt-iomuxc 0.2

It represents a new release of both packages. See the updated documentation for more information. This work will become teensy4-bsp, version 0.4.

The PR closes #95. The BSP uses [imxrt-log](https://docs.rs/imxrt-log) as its high-speed USB serial logger. `LoggingFrontend` documentation and in-tree examples demonstrate how to use the logger in hand-written `main()` and RTIC applications. We no longer need any C code for this BSP, so this co [... cut]
closed 2023-01-06 by mciantyre