mciantyre/teensy4-rs #100: EEPROM

karniv00l, 2021-05-05:
Hi there, thanks for all the great work! 

I can't seem to find any information about reading and writing from / to EEPROM. Is this something that is on the roadmap, or am I looking in the wrong place?

Thanks

mciantyre, 2021-05-05:
Thanks for checking out the project. There's no EEPROM emulation in today's BSP. You're the first to ask about it, and there hasn't been a plan to add it until someone would use it.

I'm happy to mentor you or anyone in contributing a FlexSPI driver, and EEPROM emulation based on that driver.

karniv00l, 2021-05-06:
Unfortunately I have very little experience with embedded world and with Rust, although I am interested in how one would go about developing such a functionality.

I believe this is implemented here: https://github.com/PaulStoffregen/cores/blob/master/teensy4/eeprom.c and this should be used as a reference while porting?

---

I came across this library while researching components for a future hobby project that can be written in Rust. I think Teensy is a great platform and having good tooling and libraries is essential. 

I hope I can help in some way.

mciantyre, 2021-05-16:
Yup, the EEPROM module may be a good reference implementation. We'll notice that we're using the FlexSPI peripheral throughout that code to emulate EEPROM memory. The i.MX RT reference manual describes how the peripheral works. Rather than using macros for registers, we could use the [`imxrt-ral` crate](https://imxrt-rs.github.io/imxrt-ral/imxrt_ral/index.html), which already exports [the FlexSPI registers](https://imxrt-rs.github.io/imxrt-ral/imxrt_ral/imxrt106/imxrt1062/flexspi/index.html).

If you want something working quickly, it might be easier to integrate the EEPROM code directly into the BSP, then create a Rust wrapper around the C code. c0f1fcdae2 shows what that might look like.