mciantyre/teensy4-rs #106: Add example code for generating USB key events and mouse events

mutantbob, 2021-11-27:
I would like to use my Teensy4 to generate keypresses and mouse clicks.  The C API is discussed at

https://www.pjrc.com/teensy/usb_keyboard.html

but I can not find an example in in the examples/ directory that shows how to do these sorts of things.  I would like to read an example with a name like generate_usb_keypresses.rs, or generate_usb_mouse_events.rs to provide an overview similar to the pjrc docuentation.

mciantyre, 2021-11-28:
Take a look at [the `usbd-hid` package](https://docs.rs/usbd-hid/0.5.2/usbd_hid/). The HID class lets us [send keyboard and mouse events](https://docs.rs/usbd-hid/0.5.2/usbd_hid/descriptor/index.html) to a host. `usbd-hid` requires a USB bus implementation, and we can use the USB driver in [the `imxrt-usbd` package](https://github.com/imxrt-rs/imxrt-usbd) for that implementation. [Here's an example](https://github.com/imxrt-rs/imxrt-usbd/commit/4c76b38a0f0c91a2d5f326d001b1cf59a4229d6e) of how to generate mouse events from a Teensy 4.

`imxrt-usbd` isn't part of the BSP, since the BSP has a separate USB driver for logging / serial. Nevertheless, users can pair the BSP and `imxrt-usbd` in their build if they disable the BSP's `usb-logging` feature. I'll add examples here once the existing USB driver is removed (#95).

mutantbob, 2021-12-06:
I have managed to copy the mouse code from the example you linked, and got it working in my app.

Unfortunately, I have not yet figured out how to generate keyboard events.  I have added a comment to https://github.com/twitchyliquid64/usbd-hid/issues/16 , but since the issue is closed it might not get noticed.

mciantyre, 2021-12-12:
Thanks for troubleshooting that keyboard example.

Readers who are interested in seeing a keyboard demo before it's available in this repo should see imxrt-rs/imxrt-usbd#14.

mciantyre, 2023-01-06:
Now that [imxrt-hal](https://github.com/imxrt-rs/imxrt-hal) has in-repo hardware examples, I plan to reduce the number of hardware examples maintained here. By contributing hardware examples to imxrt-hal, we'll be able to use them across multiple boards.

Teensy 4 users can still refer to the keyboard demo linked above for a demonstration. If users want that example ported to more i.MX RT boards, open an issue in imxrt-hal.

closed 2023-01-06 by mciantyre