mciantyre/teensy4-rs #107: Add example code for enabling pull-up resistor on a pin and sampling the pin.

mutantbob, 2021-12-01:
When I connect a switch to a teensy pin, the voltage floats around 0.67V instead of being pulled up to 3.3V.  If I install an external 10K pull-up resistor it works fine.

https://www.pjrc.com/teensy/td_digital.html includes an example of how to enable a pull-up resistor in C:

    pinMode(PIN_D7, INPUT_PULLUP);

I have tried using 

    let cfg = teensy4_bsp::hal::iomuxc::Config::zero().set_pullupdown(PullUpDown::Pullup27k);
    iomuxc::configure(&mut switch_pin, cfg);

but this does not appear to work like I want it to.

(some more detail and failed attempts at https://stackoverflow.com/questions/70102418/how-do-you-use-a-teensy-4-pin-via-the-teensy4-bsp-rust-crate-as-an-input-with-th )

I would like a code example that enables the Teensy4.0 pull-up resistor and reads the value

mciantyre, 2021-12-01:
See imxrt-rs/imxrt-hal#112 for a confirmed, working configuration of a **pull-down** resistor. Change the `set_pullupdown` value to your preferred pull-up resistance.

Notice the additional `set_pull_keep ` and `set_pull_keep_select` calls on the configuration; they're important for a correct configuration, and missing in our troublesome example.

---

Alternatively, use this new API to configure a PU/PD:

```rust
// Old:
// let cfg = teensy4_bsp::hal::iomuxc::Config::zero().set_pullupdown(PullUpDown::Pullup27k);

// New:
use teensy4_bsp::hal::iomuxc::{Config, PullKeeper};
let cfg = Config::zero().set_pull_keeper(Some(PullKeeper::Pullup27k));
```

This should be available with a recent update to a `teensy4-pins` dependency. If it doesn't build, run `cargo update`, and try another build. (imxrt-rs/imxrt-iomuxc#8 introduced this simplification.)

mciantyre, 2021-12-03:
Thanks again for working through this. Reach out if you hit any other issues.

I enabled [discussions](https://github.com/mciantyre/teensy4-rs/discussions/categories/q-a), which provides a Q&A, StackOverflow-like service. I'm happy to answer your questions there, too.

closed 2021-12-03 by mciantyre