mciantyre/teensy4-rs #109: Write panic message to usb

arandomliz, 2021-12-11:
This seems to be a feature request, that belongs into this repo. 

But what would be awesome to have is that the panic handler can write over USB serial the message that was provided to it. Instead of just blinking the LED. 

Is this possible?

mciantyre, 2021-12-16:
Yup, it's possible. A simple approach is to optionally integrate the `log` crate into `teensy4-panic`. The panic handler writes the message at the error level. If the BSP's USB logging feature is enabled, the panic message is written over USB serial. #111 shows a proof of concept. This approach should also work with other logging implementations, like [imxrt-uart-log](https://crates.io/crates/imxrt-uart-log).

After #95, we could consider other approaches that tightly integrate with the USB driver.

Alexander89, 2022-01-02:
I can't assign this to myself, but I will do this with the log crate. 

I will add an optional feature to activate it. If nobody complaints, RP will follow soon :smile:

mciantyre, 2023-01-06:
#95 is closed, and the idea proposed in #111 still work with the new logger. Although I'm not planning to do anything fancier than #111 right now, I'll keep this open for users who want to talk about other approaches / features.