mciantyre/teensy4-rs #113: update template

dhiltonp, 2021-12-21:
The teensy4-template uses `into_pins`, which is not in the newest API.

I guess the template targets 0.2.0, but 0.2.2 is out.

mciantyre, 2021-12-21:
Sorry, I might need some more information to understand this issue.

The test below shows that the template

- uses the latest 0.2.2 release of teensy4-bsp
- implicitly depends on [teensy4-pins v0.1.0](https://docs.rs/teensy4-pins/0.1.0/teensy4_pins/), the only release of the pins crate

`into_pins` is the recommended API to use with teensy4-pins v0.1.0. What would we use instead?

```
$ cd /tmp
$ cargo generate --git https://github.com/mciantyre/teensy4-rs-template --name hello-world
$ cd hello-world

$ cargo tree
hello-world v0.1.0 (/tmp/hello-world)
│ 
...
├── teensy4-bsp v0.2.2
│   ├── cortex-m v0.6.7 (*)
│   ├── cortex-m-rt v0.6.15 (*)
│   ├── embedded-hal v0.2.6 (*)
│   ├── imxrt-hal v0.4.5
│   │   ├── as-slice v0.1.5 (*)
│   │   ├── bitflags v1.3.2
│   │   ├── cortex-m v0.6.7 (*)
│   │   ├── embedded-hal v0.2.6 (*)
│   │   ├── imxrt-iomuxc v0.1.4
│   │   │   └── typenum v1.14.0
│   │   │   [build-dependencies]
│   │   │   └── imxrt-iomuxc-build v0.1.0
│   │   │       └── quote v1.0.10 (*)
│   │   ├── imxrt-ral v0.4.2
│   │   │   ├── bare-metal v0.2.5 (*)
│   │   │   └── cortex-m v0.6.7 (*)
│   │   ├── log v0.4.14 (*)
│   │   ├── nb v0.1.3 (*)
│   │   └── void v1.0.2
│   ├── log v0.4.14 (*)
│   ├── teensy4-fcb v0.3.0
│   │   └── imxrt-boot-gen v0.2.0
│   └── teensy4-pins v0.1.0
│       └── imxrt-iomuxc v0.1.4 (*)
└── teensy4-panic v0.2.1
```

dhiltonp, 2021-12-21:
Oops!

I wrote this ticket from memory this morning after bumping into an issue last night, I must have misremembered.

closed 2021-12-21 by dhiltonp