mciantyre/teensy4-rs #115: SPI1 don't work with pins p34, p35, p37

Alexander89, 2021-12-28:
**Describe the bug**

I'm trying to build a SPI1 interface. It appears straightforward at first glance, But the compiler refuses my code with the error:

```
 spi1_builder.build(pins.p34, pins.p35, pins.p37),
                                        ^^^^^^^^ the trait `teensy4_bsp::imxrt_hal::iomuxc::spi::Pin` is not implemented for `Pad<imxrt106x::bases::B1, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B1>>`
    = help: the following implementations were found:
             <Pad<imxrt106x::bases::AD_B0, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>> as teensy4_bsp::imxrt_hal::iomuxc::spi::Pin>
             <Pad<imxrt106x::bases::AD_B0, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B1>> as teensy4_bsp::imxrt_hal::iomuxc::spi::Pin>
             <Pad<imxrt106x::bases::AD_B0, typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>> as teensy4_bsp::imxrt_hal::iomuxc::spi::Pin>
             <Pad<imxrt106x::bases::AD_B0, typenum::uint::UTerm> as teensy4_bsp::imxrt_hal::iomuxc::spi::Pin>
           and 13 others
```

I tried all pins :see_no_evil: , but no one is working. 

**To Reproduce**
It is simple to reproduce. Use the SPI example and turn it into the SPI1

**Additional context**
The pins docs are great at the master branch, but It would be awesome If you could publish the teensy-pins crate (and add pin 48-54 as well). I like to add an SD card to my project later. I hope I'll come up with something generic and contribute to this project.

mciantyre, 2021-12-29:
Sorry, I'm having trouble reproducing this issue. I took the SPI example, and [it seems OK](https://github.com/mciantyre/teensy4-rs/commit/cc19e48e2090acb7d0dd0b0dc94c82540cb62a2c). Given your snippet, it seems that pins 34 and 35 are swapped. Could you confirm that the pins are passed into `build` as expected? Or, could you share all of your code?

Alexander89, 2021-12-29:
I'm so sorry, to open this issue and waste your time.

It works in another project. as you said, but something is wrong with my project and I still didn't figure it out. Let's dig deeper.

closed 2021-12-29 by Alexander89

mciantyre, 2021-12-29:
No problem! Reach out here, or in [discussions](https://github.com/mciantyre/teensy4-rs/discussions), if you have any other questions.