mciantyre/teensy4-rs #122: USB Host Example

saturn77, 2022-08-27:
The existing USB examples are great, but just looking for a USB Host example for SerialUSB similar to the USBHost_t36 library example found here https://github.com/PaulStoffregen/USBHost_t36/tree/master/examples/Serial

Thanks. James

mciantyre, 2022-09-04:
I'm not aware of any Rust-based USB _host_ drivers for the i.MX RT MCUs. Generally, the embedded Rust ecosystem doesn't have much for developing USB host drivers. Here's what I'm finding:

- [`usb-host`](https://docs.rs/usb-host/0.1.2/usb_host/) provides a set of traits for USB host drivers. There's a reference implementation for SAMD, and one driver implementation. A Rust-based i.MX RT host driver could support `usb-host` and define additional support for serial, HID, etc. for the ecosystem's benefit.
- The [TinyUSB](https://github.com/hathach/tinyusb) project provides a USB host API, and it supports all i.MX RTs. Using these [Rust bindings](https://docs.rs/crate/tinyusb-sys/0.1.0), it should be possible to use an i.MX RT as one of the supported USB host classes.

saturn77, 2022-09-06:
Thanks for the feedback on this. The two references that you provided  do offer paths forward. 

Of note is that the USBHost_t36 library from Stoffregen is more adaptable to being a USB Host with serial versus tinyUSB. With that said, the Rust binding to TinyUSB are showing a path forward, it is just that the details of working with serial over usb with that library will have some discovery effort. 

The question may arise why do need serial over usb and not just use a standard type FTDI or similar type connection, and the answer is that I had to develop a coprocessor type solution to an existing STM32F4 MCU that had built in USB, so the connection to it had to be as a USB Host. 

I will attempt to get the rust binding to TinyUSB to work and then progress with serialUSB method.

mciantyre, 2022-09-10:
One more approach to consider: custom Rust bindings to USBHost_t36. Pre-allocate some of the USB host objects in a C++ library, and expose a C FFI to Rust that manipulates those objects. That USB host serial example might demonstrate the objects and behaviors that Rust would need.

Personally, I've not mixed C++ and Rust, so I'm not sure of the risks with this approach. One limitation I can think of: the teensy4-bsp's runtime isn't calling static constructors. This would need to change if we must allocate USB host objects at global scope.

Best of luck. I'm happy to keep this open until we have other approaches for Rust-based USB host drivers.

saturn77, 2022-09-10:
Thank you for the very valuable insight here. I really like this approach. 

Based on your suggestion, and further looking around, I am thinking to try to make the binding with Cxx which is from here: 
https://github.com/dtolnay/cxx

This is of particular value from the perspective that the USBHost_t36 library is well proven and tested in my experience with the device side being USB Serial. 

The Cxx is developed by KDAB and is meant to be a bridge between C++ and Rust in general, and specifically for Qt GUI framework. It seems to have substantial resources and effort engaged with it. 

Thanks. James

saturn77, 2022-09-10:
Very helpful feedback !

closed 2022-09-10 by saturn77