mciantyre/teensy4-rs #123: Teensy 4.1 First set of i2c line are not accessible as there pins are not flagged as I2C pins.

Sandvoxel, 2022-09-22:
```
error[E0271]: type mismatch resolving `<Pad<imxrt106x::bases::AD_B1, typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>> as teensy4_bsp::teensy4_pins::imxrt_iomuxc::i2c::Pin>::Module == typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B1>`
   --> src\main.rs:39:30
    |
39  |     let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
    |                              ^^^^^ expected struct `typenum::uint::UInt`, found struct `typenum::uint::UTerm`
    |
    = note: expected struct `typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, _>`
               found struct `typenum::uint::UInt<typenum::uint::UTerm, _>`
note: required by a bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`
   --> C:\Users\koval\.cargo\registry\src\github.com-1ecc6299db9ec823\imxrt-hal-0.4.5\src\i2c.rs:105:23
    |
105 |         SCL: i2c::Pin<Module = M, Signal = i2c::SCL>,
    |                       ^^^^^^^^^^ required by this bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`

error[E0271]: type mismatch resolving `<Pad<imxrt106x::bases::AD_B1, typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>> as teensy4_bsp::teensy4_pins::imxrt_iomuxc::i2c::Pin>::Signal == SCL`
   --> src\main.rs:39:30
    |
39  |     let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
    |                              ^^^^^ expected enum `SCL`, found enum `SDA`
    |
note: required by a bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`
   --> C:\Users\koval\.cargo\registry\src\github.com-1ecc6299db9ec823\imxrt-hal-0.4.5\src\i2c.rs:105:35
    |
105 |         SCL: i2c::Pin<Module = M, Signal = i2c::SCL>,
    |                                   ^^^^^^^^^^^^^^^^^ required by this bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`

error[E0271]: type mismatch resolving `<Pad<imxrt106x::bases::AD_B1, typenum::uint::UTerm> as teensy4_bsp::teensy4_pins::imxrt_iomuxc::i2c::Pin>::Module == typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B1>`
   --> src\main.rs:39:30
    |
39  |     let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
    |                              ^^^^^ expected struct `typenum::uint::UInt`, found struct `typenum::uint::UTerm`
    |
    = note: expected struct `typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, _>`
               found struct `typenum::uint::UInt<typenum::uint::UTerm, _>`
note: required by a bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`
   --> C:\Users\koval\.cargo\registry\src\github.com-1ecc6299db9ec823\imxrt-hal-0.4.5\src\i2c.rs:106:23
    |
106 |         SDA: i2c::Pin<Module = M, Signal = i2c::SDA>,
    |                       ^^^^^^^^^^ required by this bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`

error[E0271]: type mismatch resolving `<Pad<imxrt106x::bases::AD_B1, typenum::uint::UTerm> as teensy4_bsp::teensy4_pins::imxrt_iomuxc::i2c::Pin>::Signal == SDA`
   --> src\main.rs:39:30
    |
39  |     let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
    |                              ^^^^^ expected enum `SDA`, found enum `SCL`
    |
note: required by a bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`
   --> C:\Users\koval\.cargo\registry\src\github.com-1ecc6299db9ec823\imxrt-hal-0.4.5\src\i2c.rs:106:35
    |
106 |         SDA: i2c::Pin<Module = M, Signal = i2c::SDA>,
    |                                   ^^^^^^^^^^^^^^^^^ required by this bound in `teensy4_bsp::imxrt_hal::i2c::Builder::<M>::build`
```

Sandvoxel, 2022-09-22:
Digging through the [imxrt-iomuxc](https://github.com/mciantyre/imxrt-iomuxc) I see that this issue has been resolved but the changes are not yet propagated to this release.

closed 2022-09-22 by Sandvoxel

stappersg, 2022-09-23:
On Thu, Sep 22, 2022 at 03:08:26PM -0700, Kyle Kovalchick wrote:
> Closed #123 as completed.
> 

Completed by what?   Please elaborate.

Groeten
Geert Stappers
-- 
Silence is hard to parse

mciantyre, 2022-09-23:
Sorry for the trouble. I reproduced the error, and I notice two issues in the example code.

1. We're still using the I2C**3** base peripheral, but we should be using the I2C**1** peripheral. The `clock` call returns a tuple, and we need to select the correct I2C instance from that result. Here's the first change that needs to happen.

    ```rust
    // From:
    let (_, _, i2c_clock, _) = periphs.i2c.clock(&mut periphs.ccm.handle, OSC, DIVIDE_3);
    // To:
    let (i2c_clock, _, _, _) = periphs.i2c.clock(&mut periphs.ccm.handle, OSC, DIVIDE_3);
    ````

2. Once we make that change, the example still doesn't compile. Pins 18 and 19 are swapped.

    ```rust
    // From:
    let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
    // To:
    let mut i2c1 = i2c_clock.build(pins.p19, pins.p18);
    ```

Tested on d7e97cb, which is using imxrt-iomuxc v0.1.1. We should have support for this peripheral and pins in today's BSP.

Sandvoxel, 2022-09-26:
> Sorry for the trouble. I reproduced the error, and I notice two issues in the example code.
> 
> 1. We're still using the I2C**3** base peripheral, but we should be using the I2C**1** peripheral. The `clock` call returns a tuple, and we need to select the correct I2C instance from that result. Here's the first change that needs to happen.
>    ```rust
>    // From:
>    let (_, _, i2c_clock, _) = periphs.i2c.clock(&mut periphs.ccm.handle, OSC, DIVIDE_3);
>    // To:
>    let (i2c_clock, _, _, _) = periphs.i2c.clock(&mut periphs.ccm.handle, OSC, DIVIDE_3);
>    ```
> 2. Once we make that change, the example still doesn't compile. Pins 18 and 19 are swapped.
>    ```rust
>    // From:
>    let mut i2c1 = i2c_clock.build(pins.p18, pins.p19);
>    // To:
>    let mut i2c1 = i2c_clock.build(pins.p19, pins.p18);
>    ```
> 
> Tested the build on [d7e97cb](https://github.com/mciantyre/teensy4-rs/commit/d7e97cb35534e4c96ddfa41e0c19a0aa85c24d91) (didn't run it), which is using imxrt-iomuxc v0.1.1. We should have support for this peripheral and pins in today's BSP.

Yes I closed the issue because I figured that out. Thanks for the timely response even thought I figured out.