mciantyre/teensy4-rs #126: USB logging: USB device not recognized

renedavoh, 2023-01-16:
**Describe the bug**
When compiling/updloading the teensy4-rs-template to Teensy 4.1 Windows doesn't recognize the COM port (since Update to 0.4 of teensy4-bsp).

**To Reproduce**
```
cargo install cargo-generate
cargo generate --git https://github.com/mciantyre/teensy4-rs-template --name hello-world
cd hello-world
cargo objcopy --release -- -O ihex hello-world.hex
```
Upload with Teensy Loader

**Additional context**
Older template using teensy4-bsp v0.3 works as expected.

mciantyre, 2023-01-17:
Thanks for trying the latest BSP, and for indicating that this issue happens on Windows. I'm interpreting "doesn't recognize the COM port" as "the USB serial logger doesn't work on Windows, because the COM port never appears." Let me know if this isn't accurate.

`teensy4-bsp` 0.4 uses [`imxrt-log`](https://docs.rs/imxrt-log/latest/imxrt_log/) for its USB serial logger. `imxrt-log` was developed against a non-Windows host OS. My first guess is that there's a USB device behavior within or beneath `imxrt-log` that Windows doesn't like.

I can't troubleshoot this by myself without a Windows host, so I'll need the community's help. If you have the time and equipment, I have two tests that might help us isolate the issue.

## imxrt-usbd serial example

Build the [`imxrt-usbd` serial example](https://github.com/imxrt-rs/imxrt-usbd) like this:

```
git clone git@github.com:imxrt-rs/imxrt-usbd.git
cd imxrt-usbd/examples/teensy4
git checkout d5a40a1
cargo objcopy --release --bin=serial -- -O ihex serial.hex
```

Upload `serial.hex` to your Teensy 4.1. At this point, you should observe a blinking LED.

Try connecting to the COM port, and enter characters. If the COM port enumerates, you should be able to submit characters and receive the same characters back.

Does this work? This evaluates a simple loopback USB serial device without `imxrt-log`.

## imxrt-hal logging example

Build the [`imxrt-hal` logging example (without RTIC)](https://github.com/imxrt-rs/imxrt-hal) like this:

```
git clone git@github.com:imxrt-rs/imxrt-hal.git
cd imxrt-hal
git checkout a15ffda
cargo objcopy --release --example=hal_logging --features=board/teensy4 --target=thumbv7em-none-eabihf -- -O ihex hal_logging.hex
```

Upload `hal_logging.hex` to your Teensy 4.1. At this point, you should observe a blinking LED.

Try connecting to the COM port. If the COM port enumerates, you should see an occasional log message with an incrementing counter.

Any luck here? This still uses `imxrt-log`, but it uses it in a slightly different way than what `teensy4-bsp` does.

renedavoh, 2023-01-17:
Same for both examples on Windows: 
- "USB device not recognized" when connected
- Device manager reports Error Code 43 ("The USB device returned an invalid USB configuration descriptor.")
- Raspberry PI did connect to the Serial port.

I compared "lsusb -v" output for this examples and old version (which uses Teensyduino library):
 - Teensyduino writes bcdUSB (USB protocol version?) 2.00
 - DescriptorWriter::device() in usb-device crate writes bcdUSB 2.10

Changing this hardcoded value in the usb_device crate made both examples work on Windows:

https://github.com/rust-embedded-community/usb-device/blob/d0b201e9d93bd8481a31d54dcb9ca75cbd32fcd5/src/descriptor.rs#L113

mciantyre, 2023-01-17:
The differing `bcdUSB` values is a helpful observation. From my understanding, when Windows (or any USB host) observes `bcdUSB > 2.00`, it can query for binary device object store (BOS) descriptors. By changing that hard-coded value from `0x10` to `0x00`, Windows shouldn't perform that BOS descriptor query.

This could indicate that something around the BOS descriptor query is failing.

mciantyre, 2023-01-17:
The differing `bcdUSB` values is a helpful observation. From my understanding, when Windows (or any USB host) observes `bcdUSB > 2.00`, it can query for binary device object store (BOS) descriptors. By changing that hard-coded value from `0x10` to `0x00`, Windows shouldn't query for BOS descriptors. This seems similar to rust-embedded-community/usb-device#35.

I don't want to rule out an issue in our own device configuration. Although there's an extra HID class in the mix, https://github.com/rust-embedded-community/usb-device/issues/102 talks about approaches to troubleshoot a CDC device against Windows. It might be easy to explore those suggestions in the `imxrt-usbd` serial example.

terrorfisch, 2023-02-09:
I tried to capture the traffic with wireshark/USBPcap to compare bcdUSB versions. USBPcap does not capture any packagesfrom the unpatched (bcdUSB 2.10) version.

psobiech, 2023-03-05:
@renedavoh Hi, as per this discussion: https://github.com/mciantyre/teensy4-rs/discussions/132#discussioncomment-5204129, could you check if this command also solves your issue (without changes to bcdUSB)?
`IMXRT_LOG_USB_BULK_MPS=512 cargo build [...]`

obim, 2023-03-09:
👍 it works!

renedavoh, 2023-03-09:
👍 With `IMXRT_LOG_USB_BULK_MPS=512` the hello-world example works on Windows.

mciantyre, 2024-06-06:
Following the recommendations in this thread, imxrt-log 0.1.2 increases the default MPS to 512.

closed 2024-06-06 by mciantyre