mciantyre/teensy4-rs #129: Must power cycle to get LED working when flashing via JTAG

cstrahan, 2023-02-08:
**Describe the bug**

When I flash one of the LED blinky examples to my Teensy 4.1 or MicroMod over JTAG, the LED only starts blinking after power cycling. 

The following program uses `defmt_rtt` to log messages via RTT, which I successfully receive (using `cargo embed`):

```rust
#![no_std]
#![no_main]

use bsp::board;
use teensy4_bsp as bsp;
use teensy4_panic as _;

use defmt_rtt as _;

use bsp::hal::gpt::ClockSource;
use bsp::hal::timer::Blocking;

const DELAY_MS: u32 = 500;

const GPT1_FREQUENCY: u32 = 1_000;
const GPT1_CLOCK_SOURCE: ClockSource = ClockSource::HighFrequencyReferenceClock;
const GPT1_DIVIDER: u32 = board::PERCLK_FREQUENCY / GPT1_FREQUENCY;

#[bsp::rt::entry]
fn main() -> ! {
    let instances = board::instances();

    let board::Resources {
        mut pins,
        mut gpt1,
        mut gpio2,
        ..
    } = board::t41(instances);

    let led = board::led(&mut gpio2, pins.p13);

    gpt1.disable();
    gpt1.set_divider(GPT1_DIVIDER);
    gpt1.set_clock_source(GPT1_CLOCK_SOURCE);

    let mut delay = Blocking::<_, GPT1_FREQUENCY>::from_gpt(gpt1);

    let mut counter: u32 = 0;
    loop {
        led.toggle();
        defmt::info!("This is a test! {}", counter);
        delay.block_ms(DELAY_MS);
        counter = counter.wrapping_add(1);
    }
}
```

But the LED doesn't blink until I power cycle the Teensy. I'm guessing some sort of board initialization issue? :thinking: 

**To Reproduce**

Kinda hard to repro without a modded Teensy :)

**Expected behavior**

LED blinky example should result in blinky LED.

**Additional context**

N/A

cstrahan, 2023-02-08:
Hm... I think this was/is only manifesting when the previous thing flashed to Teensy was a program built with the official Teensy Arduino core. After flashing and then cutting power, subsequent flashes seem to work fine. And a quick sanity checking that I can set the other pins works just fine.

Kind of curious, but I guess I'll close this for now.

mciantyre, 2023-02-08:
Good find. Here's what I guess is happening, with some context:

- The official SDK [configures fast GPIOs](https://github.com/PaulStoffregen/cores/blob/2dbb6f6c58f3d8e1272286daba3f3b094893c5b3/teensy4/startup.c#L112-L116). This means that we need to toggle the LED through the GPIO**7** registers, not the GPIO**2** registers.
- [We don't use fast GPIOs](https://docs.rs/imxrt-hal/0.5.0/imxrt_hal/gpio/index.html#todo) in this BSP / HAL. We're expecting to toggle the LED through GPIO**2** registers.

If a program built from the official SDK were flashed, those `IOMUXC_GPR` registers are set. After programming with JTAG, they're likely _still_ set until an appropriate reset can clear those `IOMUXC_GPR` register values. A power cycle must be one of those reset events.

If this is correct, something like this code (near the top of `board::prepare_resources`) might resolve the issue.

```rust
    ral::write_reg!(ral::iomuxc_gpr, instances.IOMUXC_GPR, GPR26, 0);
    ral::write_reg!(ral::iomuxc_gpr, instances.IOMUXC_GPR, GPR27, 0);
    ral::write_reg!(ral::iomuxc_gpr, instances.IOMUXC_GPR, GPR28, 0);
    ral::write_reg!(ral::iomuxc_gpr, instances.IOMUXC_GPR, GPR29, 0);
```

I expect the HalfKay bootloader handles this nuance for the majority of Teensy 4 users 😛 but it's been a while since I've tried the official SDK, so I should double-check this.