mciantyre/teensy4-rs #130: Also depending on defmt_rtt causes linker error

cstrahan, 2023-02-12:
**Describe the bug**

If I have `teensy4-bsp` as a dependency _and_ `defmt_rtt`, I run into the follow linker error:

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcWb1qH0/symbols.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1698cbrh57m8zbig.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.17hqc163sowduikf.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1c0pke542efhztpa.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1gcy1l9q970kethh.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1hjytgrzn014pvny.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1hrzdi77cga39c2c.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1o7gpd7sbhf6diez.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1svgbdrp2txfc9df.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1wiikf5tl97w8lw9.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1y7l2rxqxmu81j6w.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.1y8yj1hjx42jkhbb.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.23eso5siv5miqgeg.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2496ubo6dilz0dh0.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.24xsif871styfzdz.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2fwd7xdgzohquhah.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2mv6mmdfxcm2vodx.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2ti6y79n82ij49gd.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2vzrlktgipihdmi2.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.2wo8hke1hcvacqhr.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.31r4btohq5ecferm.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.33tw4p1tpaprbh1o.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.38d0ov15dwzqoogj.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.3gfy98u2kewwq410.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.3iewsmvvwtpzv7py.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.3kj81ct46ylel0kf.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.3ytcznt1h62or4hr.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.40ovhf3ego2s7ig3.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4194rbwvhr01zp8b.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.41pn65j27f54qux3.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.42q6yvdpvkxi5a3v.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4670j36h8p8l19y2.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4aszei5q1o6vte7p.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4lgo8xmi6jxb5zz7.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4ly43frrlq8qxr2k.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4nhzwoc8gfl86v7n.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4ov65bxvphj1wh3e.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4qb4mw64m1afmehv.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4qcvdj5yk4tr85wf.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.4yaf87072awaku4w.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.548obllg8xjin87e.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.58fsjwaubq4n7ej3.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.59orvig4dovnfi13.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.5aglzkmhce3g66wk.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.h6x7yhwp74mfihj.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.k322i9aocp18ktj.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.mfmj2tv592a8kju.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.vntr5sypm3ijwp6.rcgu.o" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/hello_teensy-74cd17b2013ebba8.y9t6s2yq1mavajr.rcgu.o" "--as-needed" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/cstrahan/src/hello-teensy/target/debug/deps" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/build/cortex-m-6a37c50e708cf516/out" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-cc7a386d27b79b4e/out" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/build/defmt-b113be18faa0503d/out" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/build/teensy4-bsp-1f714b3b36da0a35/out" "-L" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/build/imxrt-ral-c74ab446f64d079b/out" "-L" "/home/cstrahan/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-fa04cffbc8b7768e.rlib" "--start-group" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libteensy4_panic-00dcdb87c834a2e3.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libteensy4_bsp-db0abac9bda853f1.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libteensy4_fcb-f966d99042a0f0b1.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libimxrt_boot_gen-1677db8c64ed0202.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libteensy4_pins-c53fad41ea7b571c.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libimxrt_rt-f90588d32a69444c.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libimxrt_log-706459c02f6ee9f8.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libusbd_serial-e1fa295ced35fda3.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/liblog-cb9733bf151d6000.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug/deps/libdefmt-39d45ba5368247b8.rlib" "/home/cstrahan/src/hello-teensy/target/thumbv7em-none-eabihf/debug [... cut ...]
diff --git a/Cargo.toml b/Cargo.toml
index 0953c94..f2c8df8 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -48,7 +48,7 @@ features = ["imxrt1060"]
 [dependencies.imxrt-log]
 version = "0.1"
 default-features = false
-features = ["usbd", "log", "defmt"]
+#features = ["usbd", "log", "defmt"]
 
 [dependencies.imxrt-rt]
 version = "0.1"
```

**To Reproduce**

Build with both `teensy4-bsp` as a dependency _and_ `defmt_rtt`.

**Expected behavior**

Should compile successfully.

**Additional context**
:thinking:

mciantyre, 2023-02-12:
Thanks for reporting. Reproduced in the BSP template project with `defmt-rtt` version 0.4.

Both `defmt-rtt` and `imxrt-log, features = "defmt"` (version 0.1) bring in a defmt global logger. Two global loggers is disallowed, according to [the defmt docs](https://defmt.ferrous-systems.com/global-logger.html):

> ⚠️ Only a single `#[global_logger]` struct can appear in the dependency graph of an application.

It was a poor call on my part to enable `imxrt-log`'s `"defmt"` feature directly in the BSP. It violates the guidance in the next line:

> Therefore the `global_logger` should be selected at the top of the dependency graph, that is in the application crate.

I've effectively turned `teensy4-bsp` into an implicit defmt global logger. So this issue affects anyone who tries to bring in _any_ other defmt global logger.

I'm looking for non-breaking ways to fix this, but I haven't found it yet. A proper fix likely requires a new release series of the BSP with changes to the dependency / features. Additionally, I might need to re-think the `imxrt-log` common case; `"defmt"` is enabled by default.

cstrahan, 2023-02-12:
> I'm looking for non-breaking ways to fix this, but I haven't found it yet. A proper fix likely requires a new release series of the BSP with changes to the dependency / features. Additionally, I might need to re-think the imxrt-log common case; "defmt" is enabled by default.

:+1: 

Thanks again for all your work on these crates! Where I can, I'll see if I can give back in an some way.

(My apologies if it ever feels like I'm passively lobbing issue reports over the fence -- I'd normally like to take a more active role, but I'm trying to wrap up some projects while I still have my funemployment time, and I figure it's better to at least open an issue (even if hastily written and a little dry) than to hold off until I have more free time, only to probably forget about it after I've implemented some work around :sweat_smile:.)

mciantyre, 2024-05-28:
0.5.0 of teensy4-bsp drops the `imxrt-log` dependency. Users can depend on `imxrt-log` and use its `defmt` feature, or they can bring-their-own-`defmt`-logger.

closed 2024-05-28 by mciantyre