mciantyre/teensy4-rs #134: Interaction with RTIC SYSTICK handling

wyager, 2023-03-31:
Consider the following code, from an RTIC v2 app:

```rust
let systick_token = rtic_monotonics::create_systick_token!();
//Systick::start(cx.core.SYST, teensy4_bsp::EXT_SYSTICK_HZ, systick_token);
let poller = bsp::logging::log::usbd(usb, bsp::logging::Interrupts::Enabled).unwrap();
```

If I uncomment the second line, to activate the systick timer, it seems to break USB. No device even shows up under `/dev/cu....`, and my computer doesn't register any attached USB devices.

I saw in an earlier version of the teensy4-rs docs, it said that you couldn't have USB logging and systick timing at the same time, but that seems to have been removed.

Any idea what the situation might be here?

wyager, 2023-03-31:
This might actually have nothing to do with USB, as it looks like even a simple PIT led blinky is broken by uncommenting that line. Closing issue, at least for the moment

closed 2023-03-31 by wyager

wyager, 2023-03-31:
For the record, the issue seems to be the use of EXT_SYSTICK_HZ. When I switched that to 24e6, everything seems to work.

mciantyre, 2023-04-01:
SYSTCK can have [two different clock sources](https://developer.arm.com/documentation/dui0646/c/Cortex-M7-Peripherals/System-timer--SysTick/SysTick-Control-and-Status-Register?lang=en). If the clock source is set to "external clock," then `EXT_SYSTICK_HZ` is the correct frequency to use for your SYSTICK.

If I'm looking at the right code, it seems that [RTIC always uses the processor or "core" clock](https://github.com/rtic-rs/rtic/blob/064cf19265f72d7f01e0847c545e6250391a2172/rtic-monotonics/src/systick.rs#L52). `EXT_SYSTICK_HZ ` won't work in this case. If you're using the teensy4-bsp [board setup](https://docs.rs/teensy4-bsp/0.4.1/teensy4_bsp/board/index.html) to configure the clocks, then you can use [`ARM_FREQUENCY`](https://docs.rs/teensy4-bsp/0.4.1/teensy4_bsp/board/constant.ARM_FREQUENCY.html) as the clock frequency for your SYSTICK.

---

> I saw in an earlier version of the teensy4-rs docs, it said that you couldn't have USB logging and systick timing at the same time, but that seems to have been removed.

This was a design limitation of previous BSPs. Nowadays, USB logging and SYSTICK should work together. Reach out if you notice they're not jiving.